ptsouchlos/SimpleBluetoothLibrary #1: Fatal SIGSEGV 11 Error

DeveloperPaul123, 2015-04-13:
Calling simplebluetooth.endSimpleBluetooth() causes a system crash if your bluetooth socket is recieving information while you are closing the socket, resulting in a fatal segmentation error. This is probably due to trying to access the socket input stream after the socket is closed. Should try to synchronize calls to ending all the separate threads and add a flag so that when thread.cancel() is called, the input stream is never accessed (boolean flag or look at isInterrupted()).

greenspand, 2015-05-28:
It's not a bug in onDestroy method, before you write simpleBluetooth.endSimpleBluetooth(), write  simpleBluetooth.getBluetoothUtility().closeConnections(), to make sure you close everything.This way you will have a clean app close.

DeveloperPaul123, 2015-05-29:
Thank you for the comment. Have you confirmed that this works? I haven't had a chance to test it.

greenspand, 2015-05-29:
I confirm it works.I haven't received any errors. I will post the code
tomorrow.

DeveloperPaul123, 2015-06-11:
If you look at the method endSimpleBluetooth in the SimpleBluetooth class, I am already calling bluetoothUtility.closeConnections()

DavidBcn, 2016-08-05:
I confirm this is a valid work around.
I have another issue if after calling
`bluetoothUtility.closeConnections()
bluetoothUtility. endSimpleBluetooth()`

I want to do a SCAN again the library is always crashing. I even tried to put a delay before closing BT and starting it again.

`A/libc: Fatal signal 11 (SIGSEGV) at 0x00000008 (code=1)`

DeveloperPaul123, 2016-08-10:
@DavidBcn I'm not sure what work around you're alluding to. Can you clarify?

If you want to do another scan after closing the library you'll have to initialize. Also you should use the `SimpleBluetooth` class instead of the `BluetoothUtility` class to simplify things.