ptsouchlos/SimpleBluetoothLibrary #6: Multiple Connections in a bluetooth server.

DeveloperPaul123, 2015-07-30:
Support multiple connections in a bluetooth server. Should be able to implement one host and up to three client devices.

arevaldez, 2019-10-03:
Is there any update on this? Thanks!

DeveloperPaul123, 2019-10-03:
Not at this time. Is there value in doing this?

arevaldez, 2019-10-03:
I am using an android phone to try and connect to two Bluetooth devices. One device will send data to the phone and one will receive data sent from the phone. Currently, I’m only able to implement one connection at a time. From what I’ve seen, it’s because the Bluetooth socket gets closed once a connection is made. And if a new device seeks to connect, it cancels the first connection, reopens the socket, and closes it with the new device. 

I’m able to connect to both just not at the same time.

arevaldez, 2019-10-03:
I see this project hasn’t been worked on in quite a while. I’m willing to work on this enhancement.

DeveloperPaul123, 2019-10-03:
I see, so tl;dr - yes this is of value to you. 

You are correct, it's been some time since I worked on this; but maybe it's time for an update. You're definitely welcome to file a PR though. That would be much appreciated!