ptsouchlos/SimpleBluetoothLibrary #8: error: Error:Could not find property 'plugin'

kidgot, 2015-08-04:
Hi I just started to learn android, I use Android Studio.
I got error: 

Error:Could not find property 'plugin' on com.android.build.gradle.LibraryExtension_Decorated@176deea7.

How can I fix it? thanks > <

DeveloperPaul123, 2015-08-04:
I'm not sure what this is from. Can you post you build.gradle files? Also what version of gradle are you using?

kidgot, 2015-08-05:
I download the project zip file, open it with Android Studio.
first I got 
"Error:Gradle 2.4 requires Android Gradle plugin 1.2.0 (or newer)  but project is using version 1.1.0. Please use Android Gradle plugin 1.2.0 or newer.
Fix plugin version and sync project"  <= I click this ,
then I got : 
Gradle 'SimpleBluetoothLibrary-master' project refresh failed
     Error:Could not find property 'plugin' on   com.android.build.gradle.LibraryExtension_Decorated@385b9d3d.      <=@xxxxxx are different 

I try paste:  'com.github.developerpaul123.simplebluetoothlibrary:library:1.3' 
in dependencies ，the error always appear again.

DeveloperPaul123, 2015-08-11:
This should be fixed in the new version (I hope) update your build.gradle line to the following:

'com.github.developerpaul123.simplebluetoothlibrary:library:1.4@aar'

DeveloperPaul123, 2015-08-16:
Did that solve the problem?

DeveloperPaul123, 2015-10-04:
Please use the Jitpack.io releases now

closed 2015-10-04 by DeveloperPaul123