ptsouchlos/SimpleBluetoothLibrary #10: 惨

DoingLee, 2015-09-15:


DeveloperPaul123, 2015-09-15:
Could you provide more information on what exactly is going on and any errors/exceptions that are occurring? Also, posting some code snippets would be very helpful. 

Thanks

DoingLee, 2015-09-15:
I'm new for android 
All my codes were copied from your demo project (MainActivity and TestActivity).
Sometimes logcat will show error:"   7046-7046/com.sunyatsenuniversity.doing.bluetoothdemo W/Error﹕ This receiver was not registered", and other times show the error at "simpleBluetooth = new SimpleBluetooth(this, this);" : the the receiver was not unregistered

It seems that it  just simply could not receive / send message after the bluetooth was connected.

I clicked on the botton("Creat Server") in one android phone, and than clicked on the button("Connect to Server")in another phone, After the two phone paired with each other, the state was "connected".However,no mather how many times I clicked on the "Send" button after I input my messages in the EditText in one phone ,the other phone just received nothing(did not change anyting)

Is the demo working very well with your android phones?

DeveloperPaul123, 2015-09-15:
Yes from what I remember it worked decently well although I do remember there being issues with the client sending data to the server. I believe it works well when sending from the server to the client though. I'll try and look into it and see if it still works.

DeveloperPaul123, 2015-10-04:
You were right about the issue. I have fixed it in the latest commit. The newest version of the sample should be working much better.

closed 2015-10-04 by DeveloperPaul123