ptsouchlos/SimpleBluetoothLibrary #11: BluetoothAdapter﹕ getBluetoothService() called with no BluetoothManagerCallback

just4phil, 2015-09-19:
hi.
first of all: thankx for sharing this library!!
i cannot start a server...it tries for a long time but nothing happens. my logcat says: 

BluetoothAdapter﹕ getBluetoothService() called with no BluetoothManagerCallback

bye
phil

DeveloperPaul123, 2015-09-21:
Can you post any code that you are using? What types of devices are you trying to connect?

just4phil, 2015-09-21:
hi
here is my code (it is in fact just your example):
https://bitbucket.org/just4phil/djvoter/src/f8bf7afd28a91f849d0755c27b88ff87351965a0/app/src/main/java/de/philweb/djvoter/MainActivity.java?at=master&fileviewer=file-view-default

it seemd that the message "getBluetoothService() called with no BluetoothManagerCallback" is just a standard warning that shouldnt be an issue....

but i wonder why  the method "start server" doesnt work... it just runs forever...
i tested it on my mobile google nexus 4 and on my tablet google nexus 10...both the same result....

i added 3 permissions in my manifest... is that right?

bye
phil

just4phil, 2015-09-21:
i just forked/cloned your repo and tried again..... but the same result.
create server only shows "waiting..." and nothing happens

DeveloperPaul123, 2015-09-23:
That doesn't sound like it's doing anything wrong. That just means the device is waiting for a client to connect to the server. If you're just running my demo you should be able to hit create server on one device, then on the second device (also running my demo) click connect to server. Is that what you're doing or are you doing something different?

just4phil, 2015-09-23:
ok i see...
thats my fault.... now i understand :)

but i just saw that when i send a message from client to server then i cant send a message from server to client afterwards (and vice versa)

DeveloperPaul123, 2015-09-23:
Yes I believe that is correct. I was having the same issue at times with the movement of data between client and server. It's something I'm looking into since the popularity of my library is growing.

DeveloperPaul123, 2015-10-04:
I think I fixed it! Can you confirm?

DeveloperPaul123, 2015-10-04:
See the latest release for the fix.

closed 2015-10-06 by DeveloperPaul123