ptsouchlos/SimpleBluetoothLibrary #14: Receiving partial data from Arduino

mdali92, 2015-11-20:
Hi Paul,

I'm using your library for one of my project and it's working really good except for one part. I'm trying to send a message from Arduino and when I receive it via your library, I see only half or quarter of the message. I tried using BlueTerm(another bluetooth App) and I received it properly there. So, it's probably an issue on your side. Here is an example :

ORIGINAL MESSAGE (FROM ARDUINO) : 345#233#256#864$

RECEIVED BY YOUR LIBRARY : 345#233  and then 256#864# 

RECEIVED BY BLUETERM : 345#233#256#864$ (no issues)

Basically, what's happening is that I'm receiving the first part and then the second part. 

Can you look into it? Regards

DeveloperPaul123, 2015-11-20:
Hmm that is interesting. Can I see the code you're using with my library? As well as the code you have running on the Arduino? Are you using Serial.print() or Serial.println() on the arduino. Have you tried using the buffered reader for my bluetooth library with Serial.println() on the arduino. I designed that part of the library specifically with Arduino in mind.

DeveloperPaul123, 2015-11-20:
Also what is the device that you are using?

mdali92, 2015-11-21:
I'm using the sample app provided along with your library. Yes, I'm using Serial.println(). Yes, I tried using the buffer. I'm using Arduino UNO with HC-06 bluetooth module and Oneplus One Smartphone.

DeveloperPaul123, 2015-11-21:
Hmm Ok, what baud rate are you running your arduino at?

mdali92, 2015-11-21:
9600.

Ok, here is something I found. I Checked Google's official Bluetooth documentation and compared with your code : 

In your BluetoothUtility.java class , in the implementation of the ConnectedThread class, there is a run() function. There the first line of code is :

mdali92, 2015-11-21:
9600.

DeveloperPaul123, 2015-11-23:
Ok. Can I see the code you're using?

DeveloperPaul123, 2016-01-05:
Any updates on this?

mdali92, 2016-01-06:
Sorry for the delay. I was almost near my project deadline so I had to do some 'improvising' in order to get it to work. Anyways, I will send you the code which I used shortly.

Regards,
MD Azhar Ali

Date: Tue, 5 Jan 2016 05:12:59 -0800

DeveloperPaul123, 2016-03-15:
Just as an update, I still haven't received the code. Not sure if you've already sent it.

DeveloperPaul123, 2016-03-22:
See #23 for a possible solution/updates to this problem if you are still interested.

closed 2016-03-22 by DeveloperPaul123