ptsouchlos/SimpleBluetoothLibrary #15: device not avaible error

caglarispirli, 2015-12-01:
public class MainActivity extends AppCompatActivity {

    private SimpleBluetooth simpleBluetooth;
    private static final int SCAN_REQUEST = 119;
    private static final int CHOOSE_SERVER_REQUEST = 120;
    private String curMacAddress;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        Toolbar toolbar = (Toolbar) findViewById(R.id.toolbar);
        setSupportActionBar(toolbar);

        FloatingActionButton fab = (FloatingActionButton) findViewById(R.id.fab);
        fab.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View view) {
                Snackbar.make(view, "Replace with your own action", Snackbar.LENGTH_LONG)
                        .setAction("Action", null).show();
            }
        });
    }

    @Override
    protected void onResume() {
        super.onResume();
        simpleBluetooth = new SimpleBluetooth(this, this);
        simpleBluetooth.initializeSimpleBluetooth();
        simpleBluetooth.setSimpleBluetoothListener(new SimpleBluetoothListener() {

            @Override
            public void onBluetoothDataReceived(byte[] bytes, String data) {
                //read the data coming in.

            }

            @Override
            public void onDeviceConnected(BluetoothDevice device) {
                //a device is connected so you can now send stuff to it

            }

            @Override
            public void onDeviceDisconnected(BluetoothDevice device) {
                // device was disconnected so connect it again?

            }

            @Override
            public void onDiscoveryStarted() {
                Log.i("BService","discover started");

            }

            @Override
            public void onDiscoveryFinished() {
                Log.i("BService","discover finished");
            }
        });

    }

    @Override
    public boolean onCreateOptionsMenu(Menu menu) {
        // Inflate the menu; this adds items to the action bar if it is present.
        getMenuInflater().inflate(R.menu.menu_main, menu);
        return true;
    }

    @Override
    public boolean onOptionsItemSelected(MenuItem item) {
        // Handle action bar item clicks here. The action bar will
        // automatically handle clicks on the Home/Up button, so long
        // as you specify a parent activity in AndroidManifest.xml.
        int id = item.getItemId();

        if (id == R.id.action_settings) {
            //simpleBluetooth.getBluetoothUtility().enableDiscovery(300);
            simpleBluetooth.makeDiscoverable(300);
            return true;
        } else if (id == R.id.scan) {
            simpleBluetooth.scan(SCAN_REQUEST);
            return true;
        }

        return super.onOptionsItemSelected(item);
    }

    @Override
    protected void onActivityResult(int requestCode, int resultCode, Intent data) {
        super.onActivityResult(requestCode, resultCode, data);
        if(requestCode == SCAN_REQUEST || requestCode == CHOOSE_SERVER_REQUEST) {

            if(resultCode == RESULT_OK) {

                curMacAddress = data.getStringExtra(DeviceDialog.DEVICE_DIALOG_DEVICE_ADDRESS_EXTRA);
                if(requestCode == SCAN_REQUEST) {
                    Log.i("BService","try to connect : " + curMacAddress);
                    simpleBluetooth.connectToBluetoothDevice(curMacAddress);
                } else {
                    simpleBluetooth.connectToBluetoothServer(curMacAddress);
                }

            }
        }
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        simpleBluetooth.endSimpleBluetooth();
    }

}

i am trying to make little chat with using your lib. but i always get 'device not avaible' error.
i make both of device discoverable from app then start to scan from one device and try to connect other one but i get this error.

DeveloperPaul123, 2015-12-01:
I would try to pair the devices first and then connect them. I don't think i've implemented support for insecure RFCOMM sockets (i.e. connections between unpaired devices). That will come in a future release. If that doesn't help you let me know and I'll look into it.

DeveloperPaul123, 2016-01-05:
Any word on this?

closed 2016-01-13 by DeveloperPaul123