ptsouchlos/SimpleBluetoothLibrary #17: Unable to send to Arduino

CreaRo, 2015-12-10:
I am trying to send data from Android to Arduino. I am able to receive data using the sample MainActivity class in the readme. 

The code looks like this :
```java
        @Override
            public void onDeviceConnected(BluetoothDevice device) {
                //a device is connected so you can now send stuff to it
                Log.d("BA", "Connected");
                for (int i = 0; i < 100; i++) {
                    simpleBluetooth.sendData("a"); // sending ack
                    Log.d("BA", "Sent a");
                    Thread.sleep(1000);
                }
            }
```

Where am I going wrong?

DeveloperPaul123, 2015-12-10:
Can you give me some context on what exactly you're trying to send and when? The `onDeviceConnected()`  method is called only once when the device is initially connected. If possible, I would attach the sending of data to the click of a button. 

Another  potential problem is your use of `Thread.sleep(1000)`. By using this in a for loop you're blocking the thread that's trying to do the work of sending the data for 10 seconds. If you want to do it asynchronously I would use a handler. Maybe something like: 

````java 
 Handler handler = new Handler(); 
    handler.postDelayed(new Runnable() { 
         public void run() { 
              for (int i = 0; i < 100; i++) {
                    simpleBluetooth.sendData("a"); // sending a
                    Log.d("BA", "Sent a");
                    Thread.sleep(1000);
                }
         } 
    }, 1000); // send data after 1 second. 
````

DeveloperPaul123, 2016-01-04:
Any updates on this issue?

CreaRo, 2016-01-04:
I did not try after that, though it does seem like this would work.

closed 2016-01-04 by CreaRo