ptsouchlos/SimpleBluetoothLibrary #19: Alternative enableBluetooth method

ritdaw, 2016-01-11:
Currently your enableBluetooth() method will prompt the user if they wish to enable bluetooth, however seeing as the manifest has BLUETOOTH_ADMIN permission it would be useful if there was a way of enabling bluetooth without the prompt. Whilst I agree a dialog is best practice in most scenarios, there are issues in environments such as public kiosks etc where such tasks need to be completed in the background.

It would be good if there was a flag used when calling initializeSimpleBluetooth() to specify if in silent mode or not. Then any resultant calls of bluetoothUtility.enableBluetooth() would check whether to enable with  enableBluetooth() or enableBluetoothSilent().

```
    public void enableBluetooth() {
        if(this.bluetoothAdapter != null && !this.checkIfEnabled()) {
            Intent enableBluetooth = new Intent("android.bluetooth.adapter.action.REQUEST_ENABLE");
            this.mActivity.startActivityForResult(enableBluetooth, 1001);
        }

    }
```

```
    public void enableBluetoothSilent() {
        if(this.bluetoothAdapter != null && !this.checkIfEnabled()) {
            bluetoothAdapter.enable()
        }

    }
```

DeveloperPaul123, 2016-01-11:
Good point. I'll add this to the next release. Thanks!

DeveloperPaul123, 2016-01-13:
I added this feature and the `sendData(byte[] bytes)` feature in the latest commit. You can get it with: 

````java
compile 'com.github.DeveloperPaul123:SimpleBluetoothLibrary:2282a47967'
````

If you get a chance, could you test it? I haven't had time to do so. If not I'll do it sometime in the future. I'm working on a BluetoothService class and once that is done and tested I'll be drafting a new release.

ritdaw, 2016-01-13:
Great thanks, I'll give it a try today.

ritdaw, 2016-01-13:
Works perfectly, no issues as far as I can tell. Thanks.

DeveloperPaul123, 2016-01-13:
Great! Thanks for testing it out.

DeveloperPaul123, 2016-01-19:
Commit 2282a47967a066d89fa298deeae6b1ee6de4c095 closes #19

closed 2016-01-19 by DeveloperPaul123