ptsouchlos/SimpleBluetoothLibrary #20: Initialize with flag to disable toast messages

ritdaw, 2016-01-18:
It would be great if all toast messages could be disabled with a flag when initialized, in certain scenarios we do not want to display messages such as "Device not available." and "Device connected!".

DeveloperPaul123, 2016-01-18:
Yeah I recently thought the same thing. I'll do that, and I think I'm going to switch to snackbars as well.

ritdaw, 2016-01-18:
Great. If you were able to do a quick commit with a flag to disable them for now it would be much appreciated :)

DeveloperPaul123, 2016-01-19:
Ok I did a dirty commit. You can find it here: `'com.github.DeveloperPaul123:SimpleBluetoothLibrary:f20764725f'`. I haven't tested this (haven't had time today) so if you could help me out that would be great. I think I'm going to draft a new release at this point and push off the BluetoothService feature. Keep the suggestions coming I want to make this library as great and comprehensive as possible.

DeveloperPaul123, 2016-01-19:
Commit f20764725f11a1091dd578fe4247b6aac7de224a closes #20

DeveloperPaul123, 2016-01-19:
Let me know if it works if you get a chance and I'll draft the v1.5 release.

ritdaw, 2016-01-19:
Thanks, I'll try it this afternoon.

ritdaw, 2016-01-19:
Works perfectly, tried without using setShouldShowSnackbars and they show and then used setShouldShowSnackbars(false) and they don't show as intended.

it might be worth doing a release without the service in the interim.

DeveloperPaul123, 2016-01-19:
Fantastic! And yes i'm going to save that for a future release along with allowing for multiple connections on a bluetoothserver connection. Thanks for all your help.

closed 2016-01-19 by DeveloperPaul123