ptsouchlos/SimpleBluetoothLibrary #21: Update support library 23.2.0

Nanuqac, 2016-03-10:
Hi, could I ask you to update the library _com.github.afollestad.material-dialogs_ because this library refers to the resource _@drawable/abc_ic_ab_back_mtrl_am_alpha_ and this resource in support library 23.2.0 was renamed to _@drawable/abc_ic_ab_back_material_. When I tried to update support library I cannot build my app.

Thanks a lot

DeveloperPaul123, 2016-03-10:
Sure thing. I'll do that hopefully today. Sorry about that.

DeveloperPaul123, 2016-03-11:
Just released a new version with the updated library. Let me know if it fixed it.

Nanuqac, 2016-03-11:
Yes it works, thank you.

closed 2016-03-11 by Nanuqac

DeveloperPaul123, 2016-03-15:
Could I ask that you let me know what app you're developing so I can start an app list that use this library in the Readme?

Nanuqac, 2016-03-16:
Hi, I use the library for application to the warehouse where I needed to connect a barcode scanner. Unfortunately, the application is not on Google Play. Thanks for the library, has saved me a lot of time.

DeveloperPaul123, 2016-03-16:
Oh ok cool. No worries then. And great! Glad it helped.