ptsouchlos/SimpleBluetoothLibrary #22: incomplete data

smilecs, 2016-03-17:
i seem to be getting incomplete data string when passing a string concatenated with a base64 image. The base64 data is sent partial. e.g String1+":"+"String2+":"+base64image

DeveloperPaul123, 2016-03-17:
Can I see the code you are using to send the image? What exactly is missing from the message you are sending?

DeveloperPaul123, 2016-03-17:
Is the issue similar to #14 ?

smilecs, 2016-03-17:
yes its similar....i will send the code...

smilecs, 2016-03-17:
here's the code

`     simpleBluetooth.initializeSimpleBluetooth();
        simpleBluetooth.setInputStreamType(BluetoothUtility.InputStreamType.BUFFERED);
        simpleBluetooth.setSimpleBluetoothListener(new SimpleBluetoothListener() {

            @Override
            public void onBluetoothDataReceived(byte[] bytes, String data) {
                //Log.d(TAG + "incoming data", data);
               // result = result + data;
                //db.AddFriend(fp.friend());
                Log.d(TAG, data);
                try{
                    toParse = data.split(":");
                    byte[] im = Base64.decode(toParse[4], Base64.CRLF);
                    Log.d("decoder", "decoder");
                    Bitmap bit = BitmapFactory.decodeByteArray(im, 0, im.length);
                }catch (Exception e){
                    e.printStackTrace();
                }
                //friendParser fp = new friendParser(result);
                //Log.d(TAG, String.valueOf(bytes.length));
                isConnected=false;

                //read the data coming in.
            }
`

DeveloperPaul123, 2016-03-17:
Thanks for the code. 

I would try this without using the buffered stream type. When you set the stream type to buffered, the library will be looking for a newline character and will not report the results in `onBluetoothDataReceived` until a new line character is reached which is more than likely why you're losing data since I'm assuming there is no `\n` in your image data. 

Try that and update this issue with the results.

smilecs, 2016-03-17:
ok thanks very much..i'll do that

smilecs, 2016-03-18:
I tested the solution it gives the full data when i added the escape character `\n` but i noticed that while the data is being read it overrides the previous data argument on the received method. Is there a way it can send the data after all has been collected? my work around was to create a global string which i concatenate

DeveloperPaul123, 2016-03-19:
Maybe I wasn't clear in my previous post. What I meant to do is change the line: 
`simpleBluetooth.setInputStreamType(BluetoothUtility.InputStreamType.BUFFERED);` to the following:
`simpleBluetooth.setInputStreamType(BluetoothUtility.InputStreamType.NORMAL);`

DeveloperPaul123, 2016-03-21:
Did that work?

smilecs, 2016-03-21:
yes it did, sorry was suppose to close the issue. thanks for the support

SmileCs past3.com.ng

DeveloperPaul123, 2016-03-21:
No problem. Thanks for using my lib.

closed 2016-03-21 by DeveloperPaul123