ptsouchlos/SimpleBluetoothLibrary #23: Data not complete

juangdiaz, 2016-03-21:
Hi,
When my device reads the data coming in from the Arduino  the data is coming in chopped, the data im receiving comes in very fast.
Log:
03-20 21:35:18.693 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-20 21:35:18.699 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313
03-20 21:35:18.805 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9
03-20 21:35:18.818 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 0in, 231c
03-20 21:35:18.917 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-20 21:35:18.943 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9in, 229cm
03-20 21:35:19.043 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-20 21:35:19.044 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9in, 228cm
03-20 21:35:19.168 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-20 21:35:19.170 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9in, 228c
03-20 21:35:19.268 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-20 21:35:19.273 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9in, 228cm
03-20 21:35:19.394 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-20 21:35:19.397 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9in, 228cm
03-20 21:35:19.492 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-20 21:35:19.518 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 9in, 228cm

however the data should read:
03-20 21:35:18.693 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 122in, 313cm
03-20 21:35:18.805 30385-30385/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 90in, 231cm

do you know what could cause this?

also i tried using 
simpleBluetooth.setInputStreamType(BluetoothUtility.InputStreamType.BUFFERED);
and 
simpleBluetooth.setInputStreamType(BluetoothUtility.InputStreamType.NORMAL);
but is not working

im using an arduino with a proximity sensor

Thanks for the help

DeveloperPaul123, 2016-03-21:
Hello, 

Can I see your code both on the Android side as well as the Arduino side?

Thanks

juangdiaz, 2016-03-21:
the code is pretty much the same as the sample.
on the arduino side i have to ask my friend for it. since he is the one that did the arduino stuff. but all i know is the baud rate is 9600

DeveloperPaul123, 2016-03-21:
Ok, the baud rate shouldn't be a factor. If you can give me the code I can try to reproduce it on my end. 

How is the android code different from the sample? Or is it exactly the same?

juangdiaz, 2016-03-21:
Will try to get the Arduino code. or atleast the looking code.

on the android Side is exactly the same the only difference is 

```
...
public void onBluetoothDataReceived(byte[] bytes, String data) {
                //read the data coming in.
                Log.d("data: ", data);
...
```

DeveloperPaul123, 2016-03-21:
Ok thanks, I actually think I may have found an issue: 

In the BluetoothUtility.java class I have the following code in the `ConnectedThread` class: 

````java
public void run() {
            byte[] buffer;  // buffer store for the stream
            int bytes; // bytes returned from read()
            BufferedReader reader;

            if(streamType == InputStreamType.NORMAL) {
                // Keep listening to the InputStream until an exception occurs
                while (true) {
                    try {
                        bytes = mInputStream.available();
                        if(bytes > 0) {
                            buffer = new byte[bytes];
                            // Read from the InputStream
                            bytes = mInputStream.read(buffer);
                            // Send the obtained bytes to the UI activity
                            bluetoothHandler.obtainMessage(BluetoothHandler.MESSAGE_READ, bytes, -1, buffer)
                                    .sendToTarget();
                        }
                    } catch (IOException e) {
                        break;
                    }
                }
            //Buffered reader.
            } else {
                reader = new BufferedReader(new InputStreamReader(mInputStream));
                // Keep listening to the InputStream until an exception occurs
                while (true) {
                    try {
                        if(reader.ready()) {
                            String message = reader.readLine();
                            bluetoothHandler.obtainMessage(BluetoothHandler.MESSAGE_READ, -1, -1, message)
                                    .sendToTarget();
                        }
//                        bytes = mInputStream.available();
//                        if(bytes > 0) {
//                            buffer = new byte[bytes];
//                            // Read from the InputStream
//                            bytes = mInputStream.read(buffer);
//                            // Send the obtained bytes to the UI activity
//                            bluetoothHandler.obtainMessage(BluetoothHandler.MESSAGE_READ, bytes, -1, buffer)
//                                    .sendToTarget();
//                        }
                    } catch (IOException e) {
                        break;
                    }
                }
            }
````

If you notice, the object sent to the Handler is different depending on the input stream type. Then, in the BluetoothHandler in the `SimpleBluetooth` class I have this code to read the data:

````java
 private BluetoothHandler mHandler = new BluetoothHandler() {
        @Override
        public void handleMessage(Message message) {
            switch (message.what) {
                case MESSAGE_READ:
                    byte[] readBuf = (byte[]) message.obj;
                    String readMessage = new String(readBuf);
                    if(readBuf != null && readBuf.length > 0) {
                        if(mListener != null)
                            mListener.onBluetoothDataReceived(readBuf, readMessage);
                    }
                    break;
    ...
````

In the case where you using the `InputStreamType.BUFFERED`, the handler incorrectly casts the `message.obj` to `byte[]` when it is actually a `String`. This may be causing the issue. I'll try this out something this week and post the results. 

If you've forked my project, would it be possible for you to try this fix?

Thanks

juangdiaz, 2016-03-21:
Awesome!
Let me fork it this week and try to see if it solves the issue. I will let you know

DeveloperPaul123, 2016-03-21:
Great thanks. If I get a chance after work today I'll give it a try as well. 

Thanks

juangdiaz, 2016-03-22:
Wow that was fast i was about starts looking into it! but you beat me to it.
Any chance you are going to create a new tag version of it?

DeveloperPaul123, 2016-03-22:
I think I may have fixed it in the latest commit. You can try it out with this dependency:

compile 'com.github.DeveloperPaul123:SimpleBluetoothLibrary:f50e73a215'

If it does in fact fix the problem, I'll create a new release.

DeveloperPaul123, 2016-03-22:
I'm re-openning this for now until it's confirmed to be solved.

juangdiaz, 2016-03-22:
hey its not resolving the dependency for some reason

DeveloperPaul123, 2016-03-22:
Looks like jitpack is still building it. The site is also acting kind-of funky so I'd give it a minute to update.

juangdiaz, 2016-03-22:
lol ok, forgot about that part!

DeveloperPaul123, 2016-03-22:
No worries. It's usually pretty quick though, not sure what's going on.

DeveloperPaul123, 2016-03-22:
You can always check the status [here](https://jitpack.io/#DeveloperPaul123/SimpleBluetoothLibrary) and look under "Commits".

juangdiaz, 2016-03-22:
will do i will let you know as soon as i can test

DeveloperPaul123, 2016-03-22:
Sounds good. 

Thanks

juangdiaz, 2016-03-22:
`03-22 08:46:18.555 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:18.556 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 0in, 27cm
03-22 08:46:18.558 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:18.800 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:19.016 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:19.016 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c
03-22 08:46:19.168 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:19.318 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:19.319 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c
03-22 08:46:19.392 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 08:46:19.396 30672-30672/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c`

data still coming in chopped

DeveloperPaul123, 2016-03-22:
Are you using `InputStreamType.BUFFERED` ?

juangdiaz, 2016-03-22:
hahah i did the test sleepy this morning! i just look at the code and forgot to add Buffered.

I will test when i get home

DeveloperPaul123, 2016-03-22:
No worries lol. Just keep me posted.

juangdiaz, 2016-03-23:
nope didnt work
still getting 
03-22 20:58:30.907 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 2
03-22 20:58:30.965 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 2
03-22 20:58:31.040 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 2
03-22 20:58:31.208 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 2
03-22 20:58:31.267 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 3
03-22 20:58:31.348 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 4
03-22 20:58:31.506 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 6
03-22 20:58:31.570 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 8
03-22 20:58:31.721 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 20:58:31.721 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313
03-22 20:58:31.806 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 20:58:31.808 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313
03-22 20:58:31.949 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 20:58:31.950 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c
03-22 20:58:32.024 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 20:58:32.025 22962-22962/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 0in, 27cm

juangdiaz, 2016-03-23:
this works if i use https://github.com/johnhowe/BlueTerm ,but its not a lib like urs 
maybe check how they did it there.

DeveloperPaul123, 2016-03-23:
Ok thanks for getting back to me. The only difference I can find is in the `ConnectedThread` class. Specifically:

````java
public void run() {
            Log.i(TAG, "BEGIN mConnectedThread");
            byte[] buffer = new byte[1024];
            int bytes;

            // Keep listening to the InputStream while connected
            while (true) {
                try {
                    // Read from the InputStream
                    bytes = mmInStream.read(buffer);

                    mEmulatorView.write(buffer, bytes);
                    // Send the obtained bytes to the UI Activity
                    //mHandler.obtainMessage(BlueTerm.MESSAGE_READ, bytes, -1, buffer).sendToTarget();
                } catch (IOException e) {
                    Log.e(TAG, "disconnected", e);
                    connectionLost();
                    break;
                }
            }
        }
````

I'll make this change and push to master.

DeveloperPaul123, 2016-03-23:
If you get a chance, could you try this commit?

`compile 'com.github.DeveloperPaul123:SimpleBluetoothLibrary:99e7d21a0f'`

Thanks

juangdiaz, 2016-03-23:
it got worst :(

03-22 22:49:37.080 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313
                                                                                   4
                                                                                   , 313
                                                                                   4
                                                                                   , 313
                                                                                   4
03-22 22:49:37.309 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 122in, 313c
                                                                                   in, 313c
                                                                                   in, 313c
                                                                                   , 3133c
                                                                                   , 3133c
                                                                                   , 3133c
                                                                                   , 3133c
                                                                                   , 3133c
                                                                                   , 3133c
03-22 22:49:37.760 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313cc
                                                                                   n, 313cc
                                                                                   n, 313cc
03-22 22:49:37.848 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313cc
                                                                                   n, 313cc
03-22 22:49:37.988 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313cc
                                                                                   , 313cc
                                                                                   , 313cc
03-22 22:49:38.673 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 22:49:39.053 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 22:49:39.059 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c
                                                                                   n, 313c
                                                                                   n, 313c
03-22 22:49:39.202 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c
03-22 22:49:39.359 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 22in, 313c313
                                                                                   ����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
                                                                                   n, 313c313
                                                                                   ��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������� [... cut ...]
03-22 22:49:42.607 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 22:49:42.756 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 22:49:43.363 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 22:49:43.752 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 1
03-22 22:49:43.954 7164-7164/com.bluetooth.juandiaz.bluetoothconnectivity D/data:: 0in, 27cm
                                                                                   n, 27cm

DeveloperPaul123, 2016-03-23:
Wow haha. That's interesting. Looks like I'm going to have to do some more testing then. I'll have to take some time to take a closer look. Do you know if you're using `System.println()` on the Arduino side?

DeveloperPaul123, 2016-03-23:
Wait, can you try that commit w/the `InputStreamType.NORMAL` flag?

juangdiaz, 2016-03-23:
i will ask my friend will get back to you as soon as he tell me.

i tried with Normal and is giving me same junk. revert ur last commit.

juangdiaz, 2016-03-23:
thanks for the efforts its a kick ass lib and it will get better!!

juangdiaz, 2016-03-23:
ok he is going to take the arduino and tweak it a bit
i will get back to you in a few days

DeveloperPaul123, 2016-03-23:
Ok great sounds good and thanks for the kind words!

I've got one more quick fix to try. I realized that when I create the string from the bytes, I use the entire buffer, even the empty bytes, hence all the garbage. 

If this doesn't work I'll dive in a bit deeper and do some testing my self with my arduinos. 

`compile 'com.github.DeveloperPaul123:SimpleBluetoothLibrary:a9a736dc58'`

juangdiaz, 2016-03-23:
it fixed the junk char, but data is still as day one.

got n answer he is using Serial.println

im going to get him to slow down the transmission and remove duplicate data that he sends

DeveloperPaul123, 2016-07-07:
@juangdiaz Any updates on this issue? I haven't had a lot of time to test but I'm curious if you got anywhere with it?

juangdiaz, 2016-07-07:
hey @DeveloperPaul123 im on a business trip this week. but i got the improved version of the arduino but havent had a chance to test it. will do once i come back home. probably last week of july

DeveloperPaul123, 2016-07-07:
@juangdiaz Sounds good, thanks!

juangdiaz, 2016-07-31:
@DeveloperPaul123 Hey
after doing the tweaks in the arduino the data that is coming in is incomplete.
in blueterm++ the data is coming in good. however with this lib is not :/

DeveloperPaul123, 2016-08-01:
@juangdiaz Hmm, I'm not really sure what's going on then. Is the issue the same with the data being chopped?

DeveloperPaul123, 2016-08-01:
@juangdiaz Actually, one more thing I can try is thread sychronization. I'm currently not using the `synchronized` keyword in the `BluetoothUtility.java` functions, so that may be the issue.

juangdiaz, 2016-08-01:
@DeveloperPaul123 yeah the data is chopped. we reduced the times we send data from the device. but its still chopping the data.

we can try that let me know if u add the code, i can test it.

commit 6d070a5a (2016-03-22): Attempting to fix #23. Also did some overall code cleanup to avoid context leaks.