ptsouchlos/SimpleBluetoothLibrary #24: BluetoothUtility.closeConnections() crashes

DR5Guy, 2016-05-03:
BluetoothUtility.closeConnections() crashes when called with a connected BT device. The call looks like this:
` BluetoothUtility bu = simpleBluetooth.getBluetoothUtility();
bu.closeConnections();`

Tracing shows calling the close method of the Connected thread causes a problem:
`  public void cancel() {
            try {
                this.mmSocket.close();
            } catch (IOException var2) {
                ;
            }
        }`

closed 2016-05-03 by DR5Guy

DeveloperPaul123, 2016-05-03:
Was this resolved? Is that why you closed it?

DR5Guy, 2016-05-05:
No, I closed it because I wasn't 100% sure this is not because of improper activity life-cycle management. However, this may be worth taking a look.

DeveloperPaul123, 2016-05-06:
Do you mean improper life cycle management on the part of the library? Or in how you used the `BluetoothUtility` class?