ptsouchlos/SimpleBluetoothLibrary #27: Nexus 5 doesn't scan any devices

JaroGie, 2017-04-24:
On my two Nexus 5 device scanner doesn't work. Devices list on dialog is always empty. Nexus can see other devices only from settings screen, not for my app. Other phones (Samsung Galaxy S3, Huawei, OnePlus3T) scan correctly. Device pairing with Nexus work fine, however I have to do it from settings (due empty list). Could you reproduce this issue?

DeveloperPaul123, 2017-04-24:
@JaroGie Do you have any logs/screen shots to share? Are there any error messages or is the list just empty? What version of Android is the Nexus running?

I do not have any nexus devices to test with and I'm not sure I have all the hardware to do so from an emulator.

DeveloperPaul123, 2017-05-08:
@JaroGie Do you have any more info to share on this?

closed 2017-10-06 by DeveloperPaul123