ptsouchlos/SimpleBluetoothLibrary #28: can i use this library to read Glucose reading device ,

rayalarajee, 2018-03-23:


DeveloperPaul123, 2018-03-23:
Not sure. What is the protocol of your device? How does it send information?

rayalarajee, 2018-03-23:
Thank you for quick response i need to work on OneTouch Verio Flex Glucose reading device .If this repo not helps for me if possible can you please help me any reference link,so that it can help.

DeveloperPaul123, 2018-03-23:
Given that this is a off the shelf product with an app that already exists, I highly doubt you're going to be able to figure out whatever communication protocol they're using to talk to the device. You need to figure that out first. 

This library is for serial Bluetooth communication. 

For other libs see [here](https://android-arsenal.com/) and look under `Bluetooth`.

closed 2018-03-23 by DeveloperPaul123