ptsouchlos/SimpleBluetoothLibrary #29: Data Lossed with arduino

thedavekwon, 2018-08-22:
Hi, this is Dave. 
Is there anyway to prevent data loss when connected to an arduino with bluetooth?
I have connected arduino to my galaxy s6 edge. 
It works fine with the bluetooth Serial application, but I couldn't figure it out why.

DeveloperPaul123, 2018-08-28:
Can you share the code you're using on both the android and arduino side?

thedavekwon, 2018-08-28:
I was doing a project with a time limit, so I am currently using another library. 
The code was basically, sending data via bluetooth using software serial. And little modification of sample code. I tried both buffered and normal settings.

DeveloperPaul123, 2018-08-28:
I see, would you at least be willing to post the arduino code so I can replicate the issue?

DeveloperPaul123, 2018-08-30:
May I ask why you're initializing the `bluetooth` communication twice and with different baud rates?

thedavekwon, 2018-09-01:
I wanted to change baud rate, but I already soldered everything together. And I saw in the Sparkfun tutorial that I could change baud rate using this method. I  also tried with single "bluetooth.begin(115200)" too, but it gave me same result.

closed 2019-02-20 by thedavekwon